SJSURoboticsTeam/URC-Mechanical-2023 #7: Design Battery Mount

theblueone111, 2022-11-21:
The mount for the ryobi 40v batteries needs to be designed, prototyped, manufactured, and tested

theblueone111, 2023-01-13:
make the battery retionion .5mm looser all around

theblueone111, 2023-01-18:
make sure the battery is removable and not not hit the suspension mount during removal

MayaEnriquez, 2023-01-24:
rotunda drive and arm battery mount have their own flat and angled versions now, drive battery clearance issue should be fixed, arm battery needs to be changed with comm tower

MayaEnriquez, 2023-02-02:
Angled mount has heat insert slots and Velcro channel, flat mount is not changed yet.

theblueone111, 2023-02-17:
purchased battery holder from amazon to expedite process as well due to lack of connector supplier

theblueone111, 2023-02-17:
battery mounts arriving feb 22 - 24

closed 2023-07-04 by MayaEnriquez