SJSURoboticsTeam/URC-Mechanical-2023 #21: Communications Backpack

theblueone111, 2023-01-18:
A coms "backpack" needs to be designed to houseall the radio equipment and the universal board for the arm. it must be elevated above the arm battery and be an enclosure

MayaEnriquez, 2023-01-24:
First Steps 
Cad: 
Battery/mount
Transciever(Rocket 900?)
PoE connection
Access Point( after verification with mission control)
Comm Tower Case/ Wiring Setup

MayaEnriquez, 2023-02-02:
Cad done on existing components, waiting on info if we will need new ordered parts, workable platform area marked in a cad file.

MayaEnriquez, 2023-03-14:
Cad updated to reflect new notes: 2voltage regulators(wait on electrical), battery, r900, trendnet, kmetech Poe,

MayaEnriquez, 2023-03-23:
3/23 update/to do/issues:
**rotunda**
Are we allowed to cover handle since comm tower screws on rotunda after? 
Add comm tower mounting holes to arm and autonomy rotunda plates
Double check allowable overhang and arm clearance (real sits lower than cad)
Would mounting screws for items on comm tower also need to go through rotunda/ mounting plans

**autonomy specific**
LiDAR and camera FOV blocked by comm tower?
Two power supplies or not/extra voltage regulator needed?

**electrical**
Voltage regulator 12 v & 24 v, heat dissipation how?
Custom length cables/management?

**other**
Material of comm tower:
Metal + 3D printed? 
Will Utah heat be an issue

kammce, 2023-03-24:
> Voltage regulator 12 v & 24 v, heat dissipation how?

It is unlikely that the voltage regulators will need any assistance in terms of heat dissipation. But if you choose to put it in a 3D printed plastic box without air holes in order to make it dust and water resistant, then there could be some issues with heat dissipation. But the current consumption of the M9 Rocket and the AP should be quite low.

MayaEnriquez, 2023-04-20:
trendnet is replaceable with tplink as tested by Gene, 2 versions of the comm tower, notes for second version are mixed in the Autonomy Setup Github notes

MayaEnriquez, 2023-04-25:
buck ready, permission to drill holes in plates for prototype.

MayaEnriquez, 2023-04-25:
[image]
R900 Mount leaning 20 degrees leaning back to avoid the arm, if interference continues with the arm, 45 degrees is the next test. 

Power Buck ordered.

MayaEnriquez, 2023-05-04:
Tp Link Mount is roughly designed, the prototype has all the hardware, just waiting on the electrical. Second set still being looked into & wire management is not finalized. Currently no longer requires zip-ties, just 1 velcro.

MayaEnriquez, 2023-05-11:
Manufacturing Concepts:
1. Sheet Metal + risers(space for screws)
Strong in theory, sharp in practice
2. 3D Printed
Could work with ribbing + support, heat inserts & screws? 
3. Cut acrylic/plastic sheet 
Most accurate hole pattern? Also, heat inserts? Not clear
4. Holes Directly into the other rotunda plate
Screw heads would need to be countersunk properly and go up to avoid hitting I think. Con is the long manufacturing time.

MayaEnriquez, 2023-06-03:
camera mount on the pvc pipe that the r900 leans on, make it super long?