SJSURoboticsTeam/URC-Mechanical-2023 #34: Autonomy Mount

MayaEnriquez, 2023-03-15:
List to be put in Cad: Autonomy Components
Antennae + red antennae module(Needs Case Also)
IMU(Needs Case Also)
Oak D Pro Wide ( maybe Camera gimble???)
Battery Bank
Raspberry Pi(Has a Case, case has no good mounting points)
RPLIDAR
Notes: Wires for antennae seem delicate
Must fit on a rotunda with the comm tower parts but the drill battery for arm isn't needed

MayaEnriquez, 2023-03-15:
Notes from discord:
Compass Placement - Needs to be away from all electronics

GPS Placement - Better if it were away from other electronics and as high as possible

Raspberry Pi - Anywhere near the Dorito is fine as long as we have wires that can reach all the devices

Camera Placement - Needs to be in the front or middle of the rover. Also needs stability for proper sensing. (Maybe get gimbal for later?)

External Battery - Can be placed anywhere on the rover but should not be free to move.

Beacon LED - Autonomy requires an RGB LED that can be seen in bright daylight. I have been talking to Brandon about this but I don't think we've found anything yet.

acarrou, 2023-03-15:
## USB Ports we use for Autonomous Systems:
### For the RPI
- Camera | (USB A => USB C)
- GPS | (USB A => USB C)
- Lidar | (USB A => Micro USB)
- LPC | (USB A => USB C)
### USB Ports for the Power System
- camera power | (USB A => USB C)
- lidar power | (USB A => Barrel Connector)
## I2C and Power:
- IMU | data: (SCL, SDA => SCL, SDA) | power: (5V, GND)

MayaEnriquez, 2023-03-28:
Cad as of 3/28: other than antenna module, other physical components are in cad, tower needs to be made

MayaEnriquez, 2023-03-29:
Small hold(comm tower design needs to be sorted to get new available footprint)

kate-oxley, 2023-04-08:
### Stuff on top of rotunda
#### Autonomy
- RPLidar  (must have at LEAST 180° unblocked field of vision, meaning nothing can go above horizontal line in CAD)
- Raspberry Pi
- IMU (must be centered, y to front (ask Adrien which one's front))
- Cam + gimbal (gimbal needs to be bought or designed)
- Antenna + antenna module
#### Comm Tower
- Drill battery
- Power buck
- R900
- Trendnet 

### Design Notes
- Place camera/gimbal towards front
- IMU, antenna module, and power buck need cases
- antenna needs to be high (Q: how high? A: high)
- Raspberry pi has ports on almost every end, meaning it can't be blocked, will eventually need wiring clips or something, and needs to connect to things on different parts of robot (nothing moving)

- [ ] Design/buy camera gimbal (or make someone else do it, Nate or Adrien or Gene)
- [ ]  Design cases for IMU, antenna module, and power buck

kammce, 2023-04-09:
Camera Gimbal selection can be a joint thing. Meaning, it might be good for different parties to look into what type of gimbal would work and then come together to discuss which one fits the needs best. Or mechanical can simply manage this since they have information on the size and constraints of the camera intelsys is using.