SJSURoboticsTeam/URC-Mechanical-2023 #35: Bread PCB Case(Toaster)

MayaEnriquez, 2023-03-15:


MayaEnriquez, 2023-03-15:
Case for Bread PCB mountable to new arm design

MayaEnriquez, 2023-06-24:
URC 2023 > 2023 Robot > Arm > Imports > arm breakout

BadOpi, 2023-07-11:
[image]

BadOpi, 2023-07-11:
[image]

Bread Board and arm breakout mounted directly onto arm

BadOpi, 2023-07-11:
[image]

Rotunda breakout and Buck board mount prototype needs to be printed and mounted

thejonathanouyang, 2023-07-21:
Bread PCB successfully mounted to arm w/o case. Toaster Mount no longer required.

BadOpi, 2023-07-29:
[image]

Breakout and Buck Board updated mount for clearance of leg mounts