SJSURoboticsTeam/URC-Mechanical-2023 #44: End Effector(Claw)

MayaEnriquez, 2023-06-13:
Kate researched claws.
Stuff that needs to go on it: Claw, Motor, Hand Pcb(Check Slip Ring vs USB C cable), Laser pointer, whatever gears/mechanism. 
The current design needs to be extended or redone to make more space. 
The expected Degree of freedom(Pitch & Yaw range) needs to be discussed/ changed based on how the hand PCB is cabled. 
This should be changed/finished on the 2023 rover, ideally before we start on the 2024 version so we can see what to change in the future.

kate-oxley, 2023-06-13:
:)

MayaEnriquez, 2023-06-13:
Laser pointer : https://a.co/d/f2Ml6A3

kate-oxley, 2023-07-08:
### Current Issues
**Servo horn doesn’t fit because it was 3d printed**
Fix: buy one that fits
**Joints are loose**
Fix: redesign claw joints to use bushings/shoulder bolts/bearings/something that screws into something other than plastic
**Add room for accelerometer**
Note: uses usb-c cable (can’t use a slip ring)

kate-oxley, 2023-07-08:
### Hardware List (in progress)
28mm shoulder screw x4
35mm shoulder screw x2
Servo horn for FeeTech digital servo
Gear to fit servo horn & partial gears
Partial gear x2

kate-oxley, 2023-07-08:
### Just kidding now this
- add printed bushings in place of shoulder screws
- do the math
- thicken gears

kate-oxley, 2023-07-18:
# 7/15 Update
Printed & tested a laser cut/3D printed prototype
## Changes from the previous design
- Top upper links moved under the top plate to prevent interference with gears
- Doesn't use threading to keep bolts in place to allow for movement
- Clearance holes widened for moving joints
## Conclusions
#### Pros
- Actually moves
- Fully opens and closes
#### Cons
- Still relies on clamping to keep joints in place
- Gear ratio not ideal (requires too much torque)
- Doesn't have room for all of the electronics
## Changes to make
- Make gears thicker
- Decrease gear ratio
- add mounting holes for PCB board
- add space for laser diode
- Slip ring?

[image]
[image]

kammce, 2023-07-18:
Note that the accelerometer can be place on the wrist. Or if the differential wrist is used, you can put it on the feedback gear. The feedback gear is the gear on the back that looks like it doesn't do anything. The complexity of this design is that it needs a slip ring. Another option is what we did for the Rover Nessie (2nd rover) which was a board with a magnetic encoder (roll feedback) and accelerometer (pitch feedback) with a magnet on the feedback gear that can spin freely.

@kate-oxley

kate-oxley, 2023-07-19:
@kammce got it, that's very good to know. If the accelerometer's mounted on the feedback wrist, would the only wires that need to go through the slip ring be for the laser and servo? Or am I missing any components?

kate-oxley, 2024-03-29:
I'm updating the 2024 end effector issue's opening comment to better reflect the new rules, so I'm putting the 2023 one here.

## Description
The end effector is a 4 bar claw mechanism that will move small items (e.g. wrenches, hammers, etc.), transport rocks (>5 kg), and maneuver small items (e.g. type on a keyboard, tighten an allen wrench, flip switches, etc.).

## Goals/Constraints
The most relevant sections of the [rules](https://drive.google.com/file/d/1LY5VX9Psww8Bx4ub_AWMgUlajPZQ2X2A/view?usp=sharing) are as follows:
> **1.c.iii.** Objects to be retrieved in the field will consist of small lightweight hand tools (e.g.
screwdriver, hammer, wrench), small instruments (actual or simulated, e.g. a small
box), supply containers (e.g. toolbox, gasoline can), or rocks up to 5 kg in mass. All
items except the rocks will have graspable features (such as a handle) no greater than
5 cm in diameter. The maximum dimensions shall be no larger than 40 cm x 40 cm x
40 cm, but teams should expect a variety of sizes and weights.

> **1.d.i.** Rovers shall be required to perform several dexterous operations on a mock-up
> equipment system. The rover shall have to travel up to 0.25 km across relatively flat
> terrain to reach the equipment. Equipment will be between 1.5m height and the ground.
> The equipment servicing mission will involve delivering a cached science sample to a
> lander and performing maintenance on the lander. It will include the following sub-
> tasks which may be performed in any order:
> ● Pick up the cache container and transport to the lander rocket. Cache will have a
> handle at least 10 cm long and not more than 5 cm in diameter. Cache will weigh
> less than 5 kg.
> ● Open a drawer on the lander. Insert cache into a close-fitting space in the drawer,
> and close the drawer.
> ● Tighten captive screw to secure drawer. Screw will be a 5/16” Allen (hex) head.
> Teams may build the hex driver into the rover, or pick up the screwdriver provided.
> ● Undo a latch on a hinged panel of the lander and open panel.
> ● Type commands on a keyboard and follow directions on computer display. Typing
> must be correct for full points, but spelling mistakes or repeated letters may be
> corrected using the backspace/delete button or ignored for partial points.
> ● Operate a joystick (4-position, spring-return) to direct an antenna while observing a
> gauge. Gauge will be up to 20cm away from the joystick.
> ● Pick up and insert a rugged USB memory stick into a USB (type A) slot on the
> lander.
> ● Push buttons, flip switches, turn knobs.

To summarize, the end effector will interact with the following:
- Hand tools (e.g. screwdriver, hammer, wrench)
- Small instruments (e.g. a small box)
- Supply containers (e.g. toolbox, gasoline can)
- Rocks (<5 kg)
- Cache container (<10cm long, <5cm diameter)
- Drawer handle
- 5/16” allen wrench (must perform multiple rotations)
- A latch (which it will have to undo to open a hinged panel)
- Keyboard (will have to interact with individual keys without hitting the ones next to it)
- Joystick (4-position with spring return. Will require some level of precision
- USB memory stick (will be picked up and plugged in)
- Buttons
- Flip switches
- Door knobs (must be grasped and turned in place)

These are from the 2023 manual, but the guidelines shouldn't change much next year. This section will be updated when next year's manual is released

## Additional Contacts
@starphys (mission controls)
@klam20 (electronics)

## Electronics
[Raspberry Pi Zero](https://www.raspberrypi.com/products/raspberry-pi-zero-w/)
[image]
[PCB Board](https://cad.onshape.com/documents/b97908e17fca3929ddd39b66/w/d2da11b1b8072d5890f9997f/e/9e338429d6229ba6767c538d)
[image]
[Raspberry Pi Camera Board v1.3](https://uk.pi-supply.com/products/raspberry-pi-camera-board-v1-3-5mp-1080p)
[image]
[Servo](https://www.amazon.com/FEETECH-Voltage-Waterproof-Digital-Aluminium/dp/B097DKQFGM?th=1)
[image]
[Laser Diode](https://www.amazon.com/dp/B07PMJ61K3?_encoding=UTF8&ref_=cm_sw_r_cp_ud_dp_YS9MR5RZ0YQ1AVF2D850&th=1)
[image]

## Connections
Camera module --> raspberry pi via ribbon cable
Servo --> PCB board via PWM wires
Laser diode --> PCB board via power wires
PCB board --> battery (OFF CLAW) via USB-C cable

##  CAD
Access CAD [here](https://cad.onshape.com/documents/c37df51db9b15e2c88891fad/w/a9ac27298267f2e4acc75c70/e/678b541912aa5ddd8279553e)
[image]