SJSURoboticsTeam/URC-Mechanical-2023 #47: ESPCAM PCB and Enclosure

swiftrax, 2023-09-21:
Design PCB Shape and Enclosure for ESPCAM Modules to replace the RaspberryPi Cameras

swiftrax, 2023-09-24:
[Page 11](https://cdn-learn.adafruit.com/assets/assets/000/118/306/original/ov5640_datasheet.pdf?1675869959)

MayaEnriquez, 2023-11-28:
[image]

kate-oxley, 2023-11-28:
Battery CAD: https://cad.onshape.com/documents/313e58e88414a4244f7b0d63/w/17981e2a666f6c41be5abf0a/e/39c370c3da684f6fbf8131be
[image]

kate-oxley, 2024-01-06:
## Components
#### 3.7V 2000mAh Lithium Ion Battery
- [Product Link](https://www.adafruit.com/product/2011#technical-details)
- [CAD Link](https://cad.onshape.com/documents/313e58e88414a4244f7b0d63/w/17981e2a666f6c41be5abf0a/e/39c370c3da684f6fbf8131be)
#### ESPCAM Module
- [Product Link](https://www.seeedstudio.com/XIAO-ESP32S3-Sense-p-5639.html)
- [CAD Link](https://cad.onshape.com/documents/313e58e88414a4244f7b0d63/w/17981e2a666f6c41be5abf0a/e/94104771afa3fdbed158cd35)
#### Horizontal PCB Board
- [CAD Link](https://cad.onshape.com/documents/313e58e88414a4244f7b0d63/w/17981e2a666f6c41be5abf0a/e/ef64b96fcec262140792109e)
####  Vertical PCB Board
- Still being designed :(

kate-oxley, 2024-01-20:
## 1/20 Update
- The battery changed! [This](https://www.adafruit.com/product/2011#technical-details) was the old one and [this](https://www.amazon.com/EEMB-2000mAh-Battery-Rechargeable-Connector/dp/B08214DJLJ?th=1) is the new one
- The second PCB board has been finalized! The CAD is [here](https://cad.onshape.com/documents/013474a69af546f3cd9901b3/w/e3fd1047fbf118b541a6adc6/e/b0767426f575ef249fca3ff9), it's slightly wider than expected but not enough to cause any problems
- The CAD location has changed from 2024 imports to its own PS in 2023 (found [here](https://cad.onshape.com/documents/013474a69af546f3cd9901b3/w/e3fd1047fbf118b541a6adc6/e/c28bd8ebf0d8418527ccdccc))
- The overall design has changed a bit. Before each part came together individually (for ease of disassembly), but the new plan is to make the front more open so the PCBs can be soldered together (see Maya's attached drawing) 
[image]

kate-oxley, 2024-01-21:
Note: hardware is temporarily in the arm mechanical box